mac2612/retroleap #1: Retroleap not detecting device as Leappad 2

calculatortips, 2018-12-22:
Hello. Recently, I decided to install this onto my Leappad 2, in hopes of getting some new life out of it. Through some arcane witchcraft, I managed to flash the firmware successfully. On the first boot, everything worked fine; notably, the power button was detected. However, after booting the device a second time, the power button is _not_ detected -- leading me to believe that improper keybinds are being loaded, and thus, the device is not being recognized as a Leappad 2. As you can imagine, having to remove the batteries to power the device off is a bit annoying, and probably not very safe. I'm wholly inexperienced in this field, so I'm not sure where to go from here.

On a semi-related note, how do you transfer games to the device? I tried to connect to the Leappad's ethernet, but I guess 169.254.8.X isn't the subnet anymore within Retroleap, because I couldn't get nmap nor ssh to connect.

mac2612, 2019-05-30:
Hi @calculatortips, thanks for the detailed description of the problem and for collecting useful info!

I have a fix that I *think* will solve this, can you take a look at v1.2.1 and let me know if it solves the issue?

calculatortips, 2019-06-13:
@mac2612,

Apologies for the late reply. Indeed, v1.2.1 appears to have resolved the issue (checking retroarch's configuration settings shows it's loading the LeapPad 2 config file). Thank you!

closed 2019-06-13 by calculatortips