mac2612/retroleap #3: Partial Screen / Frozen LeapPad2

motown01, 2019-03-15:
I have two LeapPad2 devices. Managed to get it installed on one with no issue, the second LeapPad2 only fills a portion of the screen when booting (screenshot attached) and doesn't respond to any buttons once it loads the Retroarch menu. Tried resetting to stock and re-flashing. Same issue. Is this one not usable?
[image]

JabberwockPL, 2019-05-29:
I have the same issue, given that the resolution matches LeapsterGS, it might that it does not detect the device correctly...

JabberwockPL, 2019-05-29:
The state is not fully frozen, as it reacts to vol- and the hour changes, however, the network interface does not work.

mac2612, 2019-05-30:
Hi! Sorry for the delay, github for some reason is not notifying me on new reported issues :(

@JabberwockPL sent me an email and I took a quick look.

I'm not able to repro on my device, but I suspect what's going on is that the device detection is failing - the retroarch spawn script looks in dmesg to see if the device is a leapstergs or leappad2. If the magic line isn't present in dmesg, or the buffer has rolled off the end, it's possible the detection won't work correctly.

I put together a quick workaround for this by looking at the framebuffer size instead - if we see a leappad2-sized screen, we'll assume the device is a leappad2 and use the appropriate screen size and button layout.

I built a release as v1.2.1 with the fix - can you try this out and let me know if it works for you?

If it doesn't work, can you ssh into the device and send me the full outputs of the 'dmesg' and 'fbset' commands? That should help me understand why the detection is failing in this case.

huckleberrypie, 2019-06-03:
@mac2612 I know this is off-topic and all, but since I don't know of any other venue to ask you about things, I'd like to ask as to how was that Epic you bought last month. I'm still unable to get a suitable sponsor to host a temporary build VPS for churning out ROMs on my end, and still stuck at not getting the home screen or any other apps to show up onscreen.

statuzz, 2019-07-09:
@mac2612 HI, just following up here. I recently had 1.2.1 installed but is doing the same. Please let me know if you need anything from me to try to get this working, and thank you for your time on this!

mac2612, 2020-05-17:
Hi All, this should theoretically be fixed in the v2.0.0 alpha release.

The issue is that some of the production leappad tablets are actually a later revision than the ones I have (referred to as 'CIP' in the code IIRC). The new kernel in 2.0.0 should support the later boards; though I do not have one to test myself, I was able to test on the Ultra XDi which is a similar setup.

Can somebody give this a shot on their "new" LeapPad 2 and report back?

povilasri90, 2023-03-28:
Hey i have a leappad 2. I have followed the instructions in the readme (installed from live ubuntu 22.04), but i still get the same problem. (the picture on the screen now looks a bit different then in the original post) - but the issue is the same.  Volume up and down works but nothing else. Are there any suggestions for this ? Total noob here but maybe it is related something to the git checkout 9db71e4 part?

verbes4, 2023-06-28:
> Hey i have a leappad 2. I have followed the instructions in the readme (installed from live ubuntu 22.04), but i still get the same problem. (the picture on the screen now looks a bit different then [...]

same here, i haven't done git checkout though