mac2612/retroleap #4: No route to host

flthr, 2019-04-01:
When i do ./remote_flash.sh it says
ssh: connect to host 169.254.8.1 port 22: No route to host

JabberwockPL, 2019-05-29:
It looks like the surgeon is not loaded. I have a similar problem with virtual Linux machines on Windows - to load surgeon I need to use my vanilla Linux laptop (which is a bit inconvenient). Also make sure you restart the device right before that step.

mac2612, 2019-05-30:
Hi, it looks like your PC isn't seeing the leapfrog device in surgeon mode - can you see the surgeon mode screen on the device when you plug it in? Have you tried a different microusb cable?

Can you send me the output of 'dmesg' with the device plugged in and in surgeon mode?

JabberwockPL, 2019-05-30:
Also make sure that you start the device in the update mode right before the script is run - with my setup all my devices seem to time out after a while. And you might repeat the procedure several times - I have managed to flash them even from my Windows machine, but only after several repeats.

CODERX24, 2019-06-04:
Mine did the same too i'm running a live cd of ubuntu i need help. I also tried it on elementray OS a ubuntu os that is a operating system on my windows computer and did the same thing

mac2612, 2019-06-05:
Hi @CODERX24, what kind of USB cable are you using? Is the device showing the surgeon mode screen when you follow the instructions in the flash script?

Can you send me the output of the 'dmesg' command on your PC right after plugging in your device in surgeon mode?

CODERX24, 2019-06-07:
sorry but how do i get to the 'dmesg' and what is it?

dfawltuzr, 2019-06-24:
Hey there, I'm experiencing the same issue with my LeapPad 2. Has there been any progress on this by chance?

mac2612, 2019-06-25:
Hi @dfawltuzr,

No progress on this because I still don't have any data about what's happening in this case. Specifically, the output of the 'dmesg' terminal command as described above.

Would you be willing to collect the dmesg output as described in earlier comments and attach it here?

Thanks,
Forrest

mac2612, 2019-06-25:
@CODERX24 it's a terminal command. Open a terminal, type 'dmesg', press enter. Copy + paste the output (for example to pastebin: https://pastebin.com/) and attach a link here.

dfawltuzr, 2019-06-26:
@mac2612, I grabbed the dmesg for you and pasted it here at pastebin, hope I'm doing this right?

https://pastebin.com/zPDpsEvK

I've done this already on a LeapsterGS running Linux Mint, so my first guess may be the operating system as the one I'm trying now is GalliumOS, but I'm not sure why it would be any different as they're both Ubuntu based.

mac2612, 2019-06-27:
@dfawltuzr thanks, that works.

It looks like the device is booting into surgeon, but the network config isn't working for some reason.

Could you also get me the output of the flash script itself, and the output of the 'ifconfig' command with the device still plugged in after trying to flash? If gallium doesn't have ifconfig, you can use 'ip address' instead. You can X out your MAC addresses if you're worried about privacy.

It's probably some subtle network config thing with Gallium vs Ubuntu. I have a chromebook that I could try to reproduce this with Gallium, but it's running vanilla ChromeOS right now. It'll probably be a while before I have time to sit down and really dig into it.

In the meantime I'd definitely give it a shot on a more traditional Ubuntu-based distro like Linux Mint or regular Ubuntu. Let me know how that goes.

gknepper, 2019-08-01:
Hi All,

This is not a bug, Actually the instructions are not 100% accurate. The sshflash has some dependencies and to download them you need run the installation:

 _# install the pip for python 2_
**sudo apt install python-pip**

_# install the dependencies_
**sudo pip install pager cbf**

Then re-run the ./remote_flash.sh.

I installed using a ubuntu live and worked flawlessly. 
Good luck and regards.

panoriega, 2020-01-25:
I resolved the same issue using a ubuntu 18.10 live cd on my computer (macbook pro).  You must power on the console while pressing L + R.

JabberwockPL, 2020-01-26:
I was doing some experiments and I have reinstalled retroleap on two Leapsters and one LeapPad2 many times recently. It seems it is some hardware quirk - one Leapster get updated about two times out of ten tries - when checked over the serial connection, sometimes the NDIS over USB simply does not start. The other Leapster gets updated about eight times out of then, the LeapPad2 works every time - all with the exact same system and settings.

ryangardner, 2020-07-20:
When you boot into surgeon mode, is it just the normal screen with the leapster connected to the computer with a green background? I've been holding down the shoulder buttons and connecting it but it hasn't worked yet. 

I keep getting an error where it `can't remove surgeon_tmp.cbf` - and the error is the same if the leapster is plugged in or not, so it seems it's not getting into surgeon mode correctly. 

Is there a trick to getting it into surgeon mode other than just powering it off and then on with the shoulder buttons? 

It's connecting with vendor Id=`0f631 and productId = `001d` according to dmesg - (does the vendor / product Id change if it's actually in surgeon mode?)

jahprime, 2024-04-21:
> When you boot into surgeon mode, is it just the normal screen with the leapster connected to the computer with a green background? I've been holding down the shoulder buttons and connecting it but [...]

i had that same issue but i can be resolved by getting python 2 not 3 not 1 but 2