mac2612/retroleap #5: trying to port gmenu2x

tHubig, 2019-12-11:
This is not really an "issue", but I'm trying to "port" gmenu2x to the leapsterGS. (I say "port" because so far, i've only changed the Makefile to get it to build.) 

Do the buttons and d-pad on the LeapsterGS correspond to QWERTY keyboard keys? Which ones?

I've gotten gmenu2x to run on the leapsterGS, but only one button works, I think it is SDLK_RETURN, sdl keycode 13?

https://wiki.libsdl.org/SDLKeycodeLookup

tHubig, 2019-12-11:
Nevermind. I found retroarch.cfg.leapstergs which has the info needed.
Also, I didn't notice SDL_SetVideoMode was trying to set 320x480.

I've just about got it working! Wooohoo!

5ilver, 2019-12-24:
hey that looks great! can you push up your branch so I can try it?

tHubig, 2019-12-25:
I'm not a regular github user. I just signed up so I could ask a question or post issues if I have problems with github-hosted apps. I find it a bit oppressive and do not trust microsoft.

You can find the gmenu2x source and binary here:
http://ceratoph.freeshell.org/leapster/

A few games are included, just for fun.

5ilver, 2020-01-06:
Thanks, I got it to run. It looks very nice. We could move the conversation to gitlab or gbatmp (I think I saw you post there?) if that's better.

tHubig, 2020-01-13:
Sorry for taking so long to respond; I'm a regular in the gbatemp other handhelds section. Except, that is, when I've gone off computers for a while....

You might want to check back in the next week or so at my freeshell.org: I've  adapted a few themes and I've made a few small changes. I'm about to release a port of gmenu2x for the leappad2. (Hopefully LP2 will suck a little less that way, but it sucks pretty bad with too few keys!)

A side-effect of launching retroarch cores with scripts from gmenu2x is that you can easily have separate retroarch.cfg files for each core. You can also launch things not retroarcj cores. Separate cfgs is no big deal for LeapsterGS, but with Leappad2 (480x272) GBC looks like turds full screen and something like DOSBox or PC-98 might look better, assuming I can port it.

Anyway, rather than roll a whole new rootfs based on gmenu2x, I think it might be best to keep gmenu2x all in one folder and just tell folks to modify their /etc/init.d scripts to run it.

closed 2020-01-13 by tHubig