mac2612/retroleap #6: leappad2 board revision 0x310 not supported

tHubig, 2019-12-17:
I'm getting 320x240 and the buttons don't work like the people above.
`lf2000_gpio_init_map GPIO mapping does not support Board Revision = 0x310`

`lf2000-bl lf2000-bl: lf2000_bl_probe

lf2000-fb 0: 320 * 240 - 32 bpp (phys=86400000 virt=0xff600000 size=2457600)

lf2000-fb 0: 320 * 240 - 8 bpp (phys=86e00000 virt=0xff900000 size=2097152)`

andymcca, 2020-01-02:
Not sure if you're the same person or not, but there was a solution to this posted over at Spiffy Hacks - 

https://spiffyhacks.com/thread-1187-page-2.html

Guess this can be fixed in the main branch!

tHubig, 2020-01-13:
Yeah, I made the solution. The world of Leappad 2 hacking is too small, LOL.

closed 2020-01-13 by tHubig