mac2612/retroleap #7: Added Cores do not load - log message 'internal error'

andymcca, 2020-01-02:
Hi,
Loving this project, has breathed new life into my two Leapster GS handhelds!

I have tried manually adding up to date cores from the libretro buildbot, by using scp to copy the .so files into /usr/lib/libretro along with the other cores.  I tried with ssh to connect to the devices as described in Fuse (ZX Spectrum) and FMsx (MSX).

Neither will link in Retroleap on the device.  They show up in the 'Load Core' menu but selecting them does nothing.  In retro.log, the following messages appear -

[ERROR] Failed to open libretro core: "/usr/lib/libretro/fuse_libretro.so"
[ERROR] Error(s): /usr/lib/libretro/fuse_libretro.so: internal error

mac2612, 2020-05-17:
Hi there, which flavor of the buildbot cores are you trying to install?

You will need ARM cores, and the leapster GS does not have hardware floating point or NEON, so neither of the flavors I see on the buildbot site (armhf or armv7-neon-hf) are likely to work.

Your best bet is to find (or build) cores built for armv7l. It's relatively straightforward to build new cores by copying and modifying the package templates for the existing cores in buildroot. Actually, the cores you're trying to get are present in the retroleap buildroot tree already (see https://github.com/mac2612/retroleap/tree/master/package/libretro-fuse), they just need to be enabled in make menuconfig (though I have not tested whether they actually build correctly).

I do have a long-term plan to build more cores by default, so I'll eventually include these in the default build. In the meantime, if you happen to try it, feel free to send a pull request with the updated config that you built with or any other tweaks you had to make.

andymcca, 2021-06-13:
Hey,
Sorry I'm only just replying 18 months later!!  I managed to build a new config with some more cores and in particular got pcsx-reloaded working well with my Leapster GS by tweaking the core settings.
Unfortunately I then had to relinquish my LeapsterGS back to our youngest!!! So had to reset it etc.  
I see you've released some new Alpha versions so hopefully I can grab the Leapster back and have another play around - ill then do a pull request as you suggest with my configuration.

Many thanks for all your hard work on this!!!
Andy

andymcca, 2021-07-08:
Question for you - I've been looking at the 3DS builds of Retroarch.  Looking at this link - http://lioncash.github.io/ARMBook/about_the_3ds.html

The 3DS appears to have a similar processor to the Leapster.  I'm guessing cores from here will work? Am getting my LeapsterGS and Leappad 2 back this weekend so will try them out.

closed 2021-07-17 by andymcca