mac2612/retroleap #9: Adding roms

clownfeet, 2020-02-27:
I know this is old but hopefully someone can help.
I've flashed the leapster gs ok but just cannot add Roms.
I've tried everything for over a week.
How do i use the private key supplied with the flash when it asks for password.
Everything I do keeps coming back to the key/password.
There's absolutely no information anyway about retroleap anywhere else.
It's a shame it looks like it would be great.
I'm new to linux and the password/key thing, if anyone's still out there any help would be greatly appreciated!

trilo2600, 2020-02-27:
You will Authenticate with the key files not a password I’ll send a pic of the syntax one min

Trilo

trilo2600, 2020-02-27:
I’ll type it and tell ya as well.
After you flashes the gs change your ip to 169.254.6.1

On your ip4 usb Ethernet set the address to 169.254.6.2

This is an example of my command to ssh into the gs the path on your box will be different

ssh -I /home/trilo/Downloads/sshflash/keys/Id_rsa root@169.254.6.1

It will come up to bash 

#

To copy files and scp them over per the guide but keep the key part in there to authenticate without any password as root.

Useful for copying files to a dir you created is the wildcard as such:

scp -i /home/trilo/Downloads/sshflash/keys/Id_rsa /home/trilo/Downloads/gbaroms/*.zip root@169.254.6.1:/roms/gba

It’s also important to note your roms have to be .zip to work

Also gba bios file and others may need to be in /roms 
On the gs. Really so far I had to put gba bios file there and gameboy only so far.

If ya have more questions I’ll help ya hack it out!

Trilo

clownfeet, 2020-02-27:
Wow thank you I'll have another go today, really appreciated

trilo2600, 2020-02-27:
Sure thing come join our discord as https://discord.gg/U6ghDH

The community is small we need all the members we can get!

Trilo

clownfeet, 2020-02-27:
hi sorry to ask again.
Great instructions, I changed my IP to 169.254.6.1 and Leapter is 169.254.6.2
When I try to ssh into GS I get - Invalid ELF header
I'm using Lubuntu (try without installing) 
ssh -I /home/lubuntu/Desktop/sshflash/keys/id_rsa root@169.254.6.1
dlopen /home/lubuntu/Desktop/sshflash/keys/id_rsa failed: /home/lubuntu/Desktop/sshflash/keys/id_rsa: invalid ELF header

trilo2600, 2020-02-27:
Try setting ip4 to auto 

Connect to 169.254.6.1 That’s the up you will see it network info in retroarch.

If that auto does not work set the usb up to 169.254.6.2

Then connect again to 169.254.6.1 Your not able to change the gs ip after the flash that I know of.

Let me know how that works and join our discord as well message me there with your results

Trilo

closed 2020-03-04 by clownfeet

1991evan, 2021-03-03:
Sorry to necro this but am having same problem but i have made it to getting to the bash part then say faild to load id_rsa and asks for a password

trilo2600, 2021-03-03:
Ya have to not be specifying the key file for it to be asking for the password. Your path must be wrong show me your location of the key file where you are and what your typing?

trilo2600, 2021-03-03:
The id file is the password of the path to it is wrong it will ask for a password as it can’t find the if file. I call to the if file from what I remember in each command. I’m gonna break out my retroleap tonight to look for ya