mac2612/retroleap #10: Reference or include ports by romana0ne

Cralex, 2020-04-10:
Over on gbatemp, romana0ne ported gmenux (with a working battery meter), helpful utilities like Commander and a terminal emulator, plus some dingux games and emulators to retroleap. As someone with lots of experience using handheld emulation devices like the LDK, these inclusions really make me comfortable using it. While only the GS has enough buttons to make full use of everything, I'd love to see some sort of collaberation, or at least a reference to this software as a potential option.

Permalink for reference:
https://gbatemp.net/threads/how-to-hack-the-leapfrog-leapstergs-handheld-console.479301/#post-8932355

mac2612, 2020-05-17:
Hi Cralex, I'm very flattered that somebody decided to expand on retroleap :)

I'd be happy to take a PR from romana0ne to merge this in, if not I'll add this to the list to figure out eventually - I think the biggest issue will be the fact that some of the software he built is specific to the gp2x, which should work well on the LeapsterGS and Leapster Explorer, but I'm not sure how it will handle the different screen resolutions of the various leappad flavors (480x272, 1024x600, etc).

legop3, 2022-03-08:
any progress on this?

andymcca, 2022-11-13:
> any progress on this?

Yes - in that the LF1000 (Leapster Explorer / LeapPad 1) versions of Retroleap now run gmenunx as the frontend instead of Retroarch.