mac2612/retroleap #12: root@169.254.6.1 password?

HakanAU, 2020-07-06:
Hi there, 

Firstly, thank you for this good tutorial. I have a problem for copy content to LeapsterGS. What's the root@169.254.6.1 password? Can you give (root@169.254.6.1) password, please? When I am trying a creat folder or copy a directory or file then root@169.254.6.1 wants a password. I tried everything for copy files but it doesn't work for me copy command.

Cheers

ryangardner, 2020-07-22:
From the error, it looks like it is asking for a password because you aren't giving it the correct path for the SSH key.

From the screenshot, you have sshflash in your Desktop directory - but you are telling it to try to find the SSH key from your user directory. 

If you fix the path you are telling SCP to use to find the ssh key, it might work:
Use something like this:  `scp -i ~/Desktop/sshflash/keys/id_rsa...` 
instead of  what you are currently doing - which is : `scp -i ~/sshflash/keys/id_rsa ...`

HakanAU, 2020-08-01:
Thanks for your answer and help :) It is my mistake. I am not familiar to Linux Terminal command.  I did fix the path and everything great now.

BigBankBailey, 2024-11-07:
> From the error, it looks like it is asking for a password because you aren't giving it the correct path for the SSH key.
>
> From the screenshot, you have sshflash in your Desktop directory - but you [...]

That didn't work for me I been trying this for 2 years now still asks for a password no matter if I use Ubuntu, popos, or Windows. Please help! Idk how this has been solved when people still have issues on this topic and there's not a for sure guide that works.