mac2612/retroleap #13: `AssertionError: Path does not exist.` when trying to do sshflash

ryangardner, 2020-07-22:
I've tried on several different versions of ubuntu - and first on a vmware fusion from my mac and then on an actual linux computer running ubuntu. I've been holding down L+R when I hit power and get the image on the screen of a leapster connected to a computer. 

On any system I try to run the remote_flash from - I get the same error - which looks like it's not connecting to the surgeon properly? 

```
Please power off your leapster, hold the L + R shoulder buttons (LeapsterGS),
or right arrow + home buttons (LeapPad2), and then press power.
You should see a screen with a green background.
Press enter when you're ready to continue.
----------------------------------------------------------------
What type of system would you like to flash?
 
1. LF1000 (Didj and Leapster Explorer)
2. LF2000 (Leapster GS, LeapPad 2, LeapPad Ultra XDI)
3. LF3000 (LeapPad 3, LeapPad Platinum)
Enter choice (1 - 3)2  
Booting the Surgeon environment...
Traceback (most recent call last):
  File "make_cbf.py", line 10, in <module>
    cbf.create(mem=sys.argv[1], opath=sys.argv[3], ipath=sys.argv[2])
  File "/home/deepracer/sshflash/cbf.py", line 142, in create
    error(e)
  File "/home/deepracer/sshflash/cbf.py", line 60, in error
    assert False, e
AssertionError: Path does not exist.
Traceback (most recent call last):
  File "boot_surgeon.py", line 25, in <module>
    do_surgeon_boot(sys.argv[1])
  File "boot_surgeon.py", line 16, in do_surgeon_boot
    pager_client.upload(path)
  File "/home/deepracer/sshflash/pager.py", line 114, in upload
    self.rerror(e)
  File "/home/deepracer/sshflash/pager.py", line 64, in rerror
    assert False, 'Pager Error: %s' % e
AssertionError: Pager Error: Surgeon not found.
Done! Waiting for Surgeon to come up...rm: cannot remove 'surgeon_tmp.cbf': No such file or directory
```

Any ideas how I can troubleshoot this? I've tried with two different Leapster GS's and with two different USB cables. 
If it helps, when I hit the home button when it's on the screen after I power it on with L+R held down it gives me the info that it's:

```
U-boot: 2010.06
Build: '3.3.2.282'
...
Board ID: 0x0209
State: Connected
```

ryangardner, 2020-07-22:
I've made a _little_ progress adding some debug statements to the different pieces... 

The path `make_cbf.py` complains doesn't exist is `lf2000_surgeon_zImage` - so I'm tracking back through things to see where that's supposed to be created and why that's not working...

ryangardner, 2020-07-22:
I just renamed the surgeon_zImage to  `lf2000_surgeon_zImage` and now it's getting much farther... 

but is back to a similar issue to the other post with "no route to host"

```
----------------------------------------------------------------
What type of system would you like to flash?

1. LF1000 (Didj and Leapster Explorer)
2. LF2000 (Leapster GS, LeapPad 2, LeapPad Ultra XDI)
3. LF3000 (LeapPad 3, LeapPad Platinum)
Enter choice (1 - 3)2
Booting the Surgeon environment...
going to run make_cbf.py with memloc superhigh and surgeon_path lf2000_surgeon_zImage 
make cbf opath surgeon_tmp.cbf
make cbf ipath lf2000_surgeon_zImage
running create with ipath lf2000_surgeon_zImage
Writing CBF file.
CBF File Summary:
kernel_crc:	0x3f6c27f3
cbf_magic:	0x9abcdef0
cbf_version:	0x00000001
kernel_load:	0x00208000
kernel_jump:	0x00208000
sum_crc:	0x9ae07280
size:		0x005cac8c
Compressed: 	 True
Booting surgeon.
Done! Waiting for Surgeon to come up...Done!
ssh: connect to host 169.254.8.1 port 22: Connection timed out
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Detected Kernel partition= RFS Partition=
Flashing the kernel...ssh: connect to host 169.254.8.1 port 22: No route to host
cat: lf2000_uImage: No such file or directory
ssh: connect to host 169.254.8.1 port 22: No route to host
Done flashing the kernel!
Flashing the root filesystem...ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Writing rootfs image ( bytes)...
cat: lf2000_rootfs.tar.gz: No such file or directory
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Done flashing the root filesystem!
Done! Rebooting the host.
ssh: connect to host 169.254.8.1 port 22: No route to host
```

ryangardner, 2020-08-05:
Turns out this is an sshflash issue - the latest master branch doesn't seem to work with the 1.2.1 release of retroleap. To work around this, I reverted sshflash to an older commit:

I was able to get it to flash 1.2.1 by reverting the `sshflash` to the commit just prior to adding support for other devices. 

In my sshflash directory I went back to the older commit by running: 
`git checkout dd5abbcc8b964d3c1a88db4bef4dc51214d2d2dd`

I was then able to flash it successfully:

faradayy-cage, 2020-10-06:
i dunno if you've already figured it out, but you're supposed to put the surgeon_zImage file from the release in there, and then put your custom rootfs and uImage/zImage file in there with 'lf1000_', 'lf2000_' , or 'lf3000_' appended to the front of the filename, depending on what device you're building for.

jamesy0ung, 2021-03-20:
Same issue as OP

Shibedev12, 2023-01-11:
permanent solution. make sure you use actual linux, not WSL. i tried, failed, did it on linux and it worked.