mac2612/retroleap #14: LeapsterGS (2012) Only a few buttons work. Can't navigate menus

FatherPeeps, 2020-07-30:
I am using the Retroleap v2.0.0-alpha1.
Everything installs correctly with no issues.  When the leapstergs loads the d-pad,home button,L shoulder button, and the A button does not work.  I cant navigate to settings to change the input settings.  Is there any other way to change the input settings. 
Also I tried to flash the retroleap v1.2.1 but the newer version of sshflash will not let me do it. Is their a way to get the older version of sshflash?
Thanks

ryangardner, 2020-08-04:
Oh, maybe that was my issue in #13 was the sshflash version - do you see the same errors I see when you try to use sshflash to flash 1.2.1? 

I can probably figure out how to get an older version of sshflash if that's actually what the problem is. You can probably just git pull an earlier commit of it.

ryangardner, 2020-08-05:
I was able to get it to flash 1.2.1 by reverting the `sshflash` to the commit just prior to adding support for other devices. 

To do that, go to your sshflash directory and tell git to revert to that older commit by running: 
`git checkout dd5abbcc8b964d3c1a88db4bef4dc51214d2d2dd`

It took me a couple of tries to get things to work after that, but it did eventually work.