mac2612/retroleap #17: says failed to mount and says no route to host

ll3macorn, 2021-01-04:


jamesy0ung, 2021-03-20:
Same issue here

Log included

`./remote_flash.sh 
Leapster flash utility - installs a custom OS on your leapster!

WARNING! This utility will ERASE the stock leapster OS and any other
data on the device. The device can be restored to stock settings using
the LeapFrog Connect app. Note that flashing your device will likely
VOID YOUR WARRANTY! Proceed at your own risk.

Please power off your leapster, hold the L + R shoulder buttons (LeapsterGS), 
or right arrow + home buttons (LeapPad2), and then press power.
You should see a screen with a green background.
Press enter when you're ready to continue.
Booting the Surgeon environment...
Writing CBF file.
CBF File Summary:
kernel_crc:	0x3f6c27f3
cbf_magic:	0x9abcdef0
cbf_version:	0x00000001
kernel_load:	0x00208000
kernel_jump:	0x00208000
sum_crc:	0x9ae07280
size:		0x005cac8c
Compressed: 	 True
Traceback (most recent call last):
  File "util.py", line 20, in <module>
    do_surgeon_boot('surgeon_wrap.cbf')
  File "util.py", line 15, in do_surgeon_boot
    pager_client.upload(path)
  File "/home/james/Desktop/sshflash/pager.py", line 114, in upload
    self.rerror(e)
  File "/home/james/Desktop/sshflash/pager.py", line 64, in rerror
    assert False, 'Pager Error: %s' % e
AssertionError: Pager Error: Mount Error: Device not found.
Waiting for Surgeon to come up...
Flashing the kernel...
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Flashing the root filesystem...
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Writing rootfs image (29130752 bytes)...
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Done! Rebooting the host.
ssh: connect to host 169.254.8.1 port 22: No route to host
`

andymcca, 2021-07-17:
This suggests that you aren't getting a network connection between the Leapfrog device and your machine.  When plugging in the device via USB, your machine should create a new Ethernet connection.  You should also wait until remote_flash.sh instructs you to power on the device before doing so as I've found it can time out otherwise

Please give more info i.e. OS you are using, Leapfrog device etc.

Aliliche, 2023-02-20:
> This suggests that you aren't getting a network connection between the Leapfrog device and your machine. When plugging in the device via USB, your machine should create a new Ethernet connection. [...]

I have the same issue on debian 11, the Leapster is recognized by the kernel, i can see it with `dmesg`  but can't be mounted. 
There are no node in  /dev , and `lsblk`  shows nothing about Leaspster. 
I think  the device have  to be mounted manually.