mac2612/retroleap #23: Surgeon reported USB Class / Subclass causes Windows 10 to use wrong network driver

andymcca, 2021-07-21:
As per the title.

For the Surgeon supplied with Retroleap 1.2.1, device USB\Class_02\Class_02 is presented, and Windows installs a generic 'USB Serial Adapter' driver.

For Retroleap 2.0.0 Alpha, Windows seemingly detects the device correctly and installs the usbnet.inf driver.  Unfortunately, the device then fails to start (Code 10).

andymcca, 2021-07-21:
A workaround I have found is to use the version of Surgeon included with Retroleap 2.0.0 , then manually change the driver in Device Manager to use this one from Datalogic - https://github.com/datalogic/usblan.

I then have to manually set a static IP on the adapter (169.254.8.10), after which I can SSH to the Surgeon-booted device (in my case, a LeapPad 2).

I haven't looked at Surgeon yet, but is it relatively easy to change the USB Class \ Subclass presented?  If so, this could be modified so that devices will work 'out of the box' with Windows without the need for the Datalogic driver.  I'm working on a fork of sshflash specifically geared towards Windows, so would be good to have some assistance with this!!

Thanks

mistermuki, 2021-07-22:
No promises but I can try taking a look at it. Are you the creator of the repository on an alt account?

andymcca, 2021-07-22:
@mistermuki that would be great, even just taking a look would be awesome, thank you! 

No I'm not the creator of this repository, but I have forked mac2612 / sshflash as mentioned above to create sshflash-win.  I've conversed with mac2612 before on a separate issue and he is relatively active here so hopefully he will comment/give some guidance at some point.  In the meantime if you manage to take a look and find anything out that would be great :-)

andymcca, 2021-07-29:
I've flashed a Leapster Explorer in the last few days, and while Windows 10 detected and loaded the Generic EEM driver successfully with no Code 10 error, connection to Surgeon still didn't work!  I can confirm though that the Datalogic driver does work, so this workaround will now work with LF1000 and LF2000 devices at least.

mistermuki, 2023-01-08:
Hey! Forgot to get back to you on this and I got pretty busy. I've noticed you've been doing a lot of work on this project.

If you're still working on it, perhaps we could do something more in line with reviving the project? Either way, I'd love to communicate and toss the idea about. 

If you're on Discord, you can message me at mister muki#7878.
Otherwise, I can give you my e-mail or something.

andymcca, 2023-01-08:
@mistermuki Hello!  Nice to hear from you.  Yes am still quite active messing about with the Leapster devices, but yes I'm also quite busy these days so can't spend as much time as I'd like on it!  I've been chatting a bit with @mac2612 (creator of this repo) on the openlf Discord group and he is continuing to work on retroleap as and when he can, but like all of us real life gets in the way!

Thanks for your Discord handle, I need to get Discord set up on my Laptop then I'll send you a bit of a summary of where things are and what I'm working on.  In terms of this particular issue (USB Class / Subclass reporting), it's down the the usbether gadget used to configure the USB port on the Leappad/Leapster.  I recently sent mac a couple of possible solutions to either get Windows working natively, or to use the Datalogic driver more easily (i.e. not overriding Windows driver choice), so I'll forward these on to you.

All the best and thanks for coming back to me! :-)

mistermuki, 2023-01-21:
@andymcca By any chance, do you have an invite to the OpenLF Discord? I'd love to join in on the conversations.