mac2612/retroleap #26: Need help adding roms to Leapster Learning System, Retroleap

phanicowner, 2022-06-26:
Hi so, I finally managed to get retroleap on to the leapfrog learning system but I cannot seem to add roms to it, I need this done soon because August 4th will be my nephew's 5th birthday and I want games setup for him already for him to play. I followed the tutorial provided by Andy's Retro Tech! https://www.youtube.com/channel/UCSlLFHMDa5eIohpw5bWBFvQ about how to get it on there via Windows 10 method rather linux. but, adding roms appears to be the difficult part.

I read the comments under one of the tutorials for the leapster learning system retroleap setup and he responded to one user with a similar issue with the following: 

"the keys are included with sshflash-win, not Retroleap. They are in the folder named keys.  If you use them with WinSCP and the instructions for linux SCP in the Retroleap Readme, you should (hopefully) be able to figure it out" and, "I think you have to set up a new site in WinSCP, configure it to use an RSA key to connect, then I think for the login it's username root with no password.  I'll try it myself later today if I have time". 

So this lies with one question, how do I set this all up? I have WinSCP but been struggling and struggling with it. im just so lost if anyone can help that would be amazing. :-D

mistermuki, 2022-06-26:
Can you connect to your newly flashed device at all? (eg. through SSH)

phanicowner, 2022-06-26:
Well I tried to connect to it before thru cmd but I think it said something
about connection timing out.

heyitzrare, 2022-06-26:
Okay, so here's my take: using Windows to do this is so much of a pain, it's frankly more worth it to grab a flash drive, throw an Ubuntu livecd on it, and use its terminal to do this entire process. Ask me how I found that out :P

phanicowner, 2022-06-26:
Ooo didn't know you could do that. Alright, ill do that for I have a usb
drive and such but afterwards what do I do after that? Can you give me step
by step on how to add the roms perfectly unto it? I read the readme and was
confused 😕

phanicowner, 2022-06-26:
> Also how can I check how much space is on the leapster learning system for
> I basically downloaded a huge load of GBA games for it.

Actually found a pretty good linux tutorial on youtube I think I'm good
now. thank you nonetheless, and if I have any issues I'll let you guys know.

phanicowner, 2022-06-27:
Hi,
So update: I managed to finally add roms and such I should've thought to run Linux a long time ago~ thank you but there is one more thing I need help with. I want to add GBA games unto it YET, when reading on how to it mentions something about, "you need to put your GBA bios at /roms/gba_bios.bin on the device (see 'Adding ROMs')" the thing is how do I do that? I read it and I'm completely lost.

heyitzrare, 2022-06-27:
Yeah, you'll need a GBA BIOS - I have one in my library if you're having trouble! Do you have a Discord?

heyitzrare, 2022-06-27:
I just sent a request, gotta keep the username secret because there's some people goin after me 😔

phanicowner, 2022-06-27:
> I just sent a request, gotta keep my username secret because there's some
> people goin after me 😔
>
> EDIT: no I didn't because I couldn't, can you turn on friend requests real
> quick?

Done.

heyitzrare, 2022-06-27:
Sent a request

phanicowner, 2022-06-28:
If someone can tell me how to add a gba_bios.bin? Thnx

phanicowner, 2022-06-28:
I have the gba_bios.bin file that Trey given me. but the thing is, I have
no idea how to input this well like what command. like on the readme it
says, "

you need to put your GBA bios at /roms/gba_bios.bin on the device (see
'Adding ROMs')"

The thing is how? I did, ssh -i ~/sshflash/keys/id_rsa
***@***.*** 'mkdir /roms/gba_bios.bin' and it made a dir on the
system called gba_bios.bin but I still can't even add GBA games unto
it.

trilo2600, 2022-06-28:
Yes google that Gameboy bios file and just put it in the folder that your .gba roms are in the core you choose will see the bios file and it should load. I have to break out my leap to try it however it’s been in the basement. I’m glad ya got the Roma on there with Linux I’m sure it was mad easier than windows!

heyitzrare, 2022-06-28:
> Yes google that Gameboy bios file and just put it in the folder that your .gba roms are in the core you choose will see the bios file and it should load.

I gave them a BIOS file, in case it wasn't obvious lol

andymcca, 2022-10-10:
Tutorial video here showing how to add files (i.e. Emulators/ROMS etc) to a Leapster/Leappad runing Retroleap, using WinSCP -
https://www.youtube.com/watch?v=o1QBUB5LbjY

phanicowner, 2022-10-11:
Also how can I check how much space is on the leapster learning system for
I basically downloaded a huge load of GBA games for it.

andymcca, 2022-12-15:
@phanicowner if you're using WinSCP I think you can open a console session to the device and then issue a 'df -h' command which should give you the available space on the device

phanicowner, 2022-12-15:
actually, realized that and figured it out a long long time ago. and I was
able to add the best titles and such, and just gave it to my neeses and
nephews for them to mess around with everytime they came over here. so, I'm
good and about finished with turning the learning system into a retro
handheld. it's jam-packed with games and such so I'm gucci.

andymcca, 2022-12-15:
@phanicowner sorry it took so long to reply to your original query but pleased you've got everything sorted out.  :-)

phanicowner, 2022-12-15:
No problem, thanks!