mac2612/retroleap #35: Random freezing whilst in emulators and Retroarch cores (LF1000/Leapster Explorer)

andymcca, 2022-11-25:
I'm logging this here as an issue as during my work on Retroleap and associated emulators/cores, I've noticed a pattern of random freezes occurring on the Leapster Explorer.  At first I thought this was just my Explorer playing up but I have had it confirmed by other users on the LF Discord group that they are encountering the same.  These have taken the form of -

- Freezing in PocketSNES standalone
- Freezing in gpSP standalone test version (sometimes recoverable after code modification to minimise SDL_CONDWAIT loops)
- Freezing in RetroArch/Fuse core (sometimes recoverable i.e. unresponsive for 3-4 minutes, then will come back on)
- Freezing in RetroArch/gpSP core (in Sega Arcade Gallery/Outrun when sound is particularly choppy)
- Freezing in RetroArch/Handy (in Toki not long after starting the game)

As overclocking has generally been present in the above scenarios, this was the first assumption as the reason.  But my investigations so far are actually pointing to a problem with alsa sound, and I'm wondering if it could be down to a faulty kernel driver.

In RetroArch I've found the problem goes away if you switch from the alsa driver, to the alsathread driver.  I assume the latter runs audio on a separate thread, meaning if there is a problem that thread will either be restarted/recreated without affecting the rest of the applications.  Similarly, I largely fixed the issue on standalone gpSP by removing a couple of while loops which were waiting on SDL_Mutex availability relating to sound.  I noticed after doing this that I would get freezes in game but these would last seconds rather than the infinity as described above!

I will continue my investigations into this and post any updates as they arise.  Feel free to contribute with reports or your own investigations!

pa2ngehe, 2022-12-13:
> I'm logging this here as an issue as during my work on Retroleap and associated emulators/cores, I've noticed a pattern of random freezes occurring on the Leapster Explorer. At first I thought this [...]

can wait to testing out your standalone gpSP on my leapster explorer.
Cheers...

andymcca, 2022-12-15:
@pa2ngehe check out my latest release for Retroleap / LP Explorer here -

https://github.com/andymcca/retroleap/releases/tag/retroarch-v1.13.0-for-retroleap

I've updated it in the last few days with a better Gfx driver.  gpSP runs at least as good as the standalone version, and has the benefit of around 10 years worth of updates over the standalone version!

andymcca, 2022-12-15:
@pa2ngehe if you want to try the standalone gpSP for the Explorer, my last release is here -
https://github.com/andymcca/gpsp-retroleap/releases/tag/v04

pa2ngehe, 2022-12-20:
first, think i want to ask is : is this standalone gpSP newer than your gpSP on RA add-on you forked lastday i installed?

i see on that page, readme is too short in explanation in section "how to install", i can't find it there.

andymcca, 2022-12-20:
No it isn't - it's forked from the notaz version of gpSP, which last had a code commit in 2013.  I forked/updated it following a conversation with Ramonez on the Openlf Discord, as he asked for help getting the controls working on this version.  I found a few other problems with it which I fixed, but really in my opinion the RetroArch version is much better, as this has been worked on much more in recent years.