mac2612/retroleap #38: sshflash is asking for password again.

larson9999, 2023-01-01:
I'm trying to update firmware again on the a couple of new devices and am having the issue were it's asking for the password again. This is the same system from which I successfully updated several devices a few months ago.  After getting the issue with new devices I tried on devices which I'd already installed retroleap.  It's asking for passwords on those too.  

I'm running ubuntu mate 22.04.
OpenSSH_8.9p1 Ubuntu-3ubuntu0.1, OpenSSL 3.0.2 15 Mar 2022

In case it matters, I'm able to ssh into 169.254.6.1 with no issues on the devices which already have retroleap installed.

What could I be doing wrong?

larson9999, 2023-01-02:
I'm a total idiot when it comes to network stuff so take this with a grain of salt.  I solved this by creating a file named 'config' in my .ssh directory with this line.

PubkeyAcceptedKeyTypes +ssh-rsa

Apparently and upgrade of ssh blocks this by default.  Figuring I don't want it permantly, I'll remove it after I update the devices I want to update.

closed 2023-01-02 by larson9999

brodieman23, 2024-10-20:
> I'm a total idiot when it comes to network stuff so take this with a grain of salt. I solved this by creating a file named 'config' in my .ssh directory with this line.
> 
> PubkeyAcceptedKeyTypes +ssh-rsa
> 
> Apparently an upgrade of ssh blocks this by default. Figuring I don't want it permanently, I'll remove it after I update the devices I want to update.

Absolute legend. Thankyou my friend! This should be part of the official documentation!