mac2612/retroleap #41: Question. How To Add Cores?

Shibedev12, 2023-01-11:
So If I wanted to play pokemon blue, how would i go about running gb games on retroleap?

andymcca, 2023-01-17:
You have several options depending on your Leapfrog device.

In the stock releases of retroleap for LF2000 (Leapfrog Leappad 2 and Leapster GS Explorer), the gpsp core is included in Retroarch for playing GBA games.  For LF1000 (Leappad 1 and Leapster Explorer), ohboy is included for running Gameboy/Gameboy Color games.

If you check out my repos you'll find additional cores, an overclocked version of retroleap for Leapster GS Explorer and an add-on pack for LF1000 retroleap (specifically the Leapster Explorer) which adds Retroarch/gpsp.

You also have the option of building retroleap yourself from the github sources, during which you can specify which cores/emulators you'd like to build (instructions in the retroleap readme).

Shibedev12, 2023-01-19:
thanks! i forgor that that was a thing. now i can play faceball 2k!

closed 2023-01-19 by Shibedev12