mac2612/retroleap #44: Seeing request for root@169.254.8.1 password when connecting.

BullieCaSa, 2023-02-04:
Following the readme for Retroleap v2.0.0-alpha4, but running on Ubuntu 22.04.1 LTS with python 2 installed. Trying to flash LF2000 (LeapsterGS) Everything is good up until the request for a password during ssh. Not sure of this is because SSH keys are in the process of being removed from sshflash, but I also tried using the older version with no success. Thanks!

`You should see a screen with a green background.
Press enter when you're ready to continue.

What type of system would you like to flash?

1. LF1000-Didj (Didj with EmeraldBoot)
2. LF1000 (Leapster Explorer)
3. LF2000 (Leapster GS, LeapPad 2, LeapPad Ultra XDI)
4. LF3000 (LeapPad 3, LeapPad Platinum)
Enter choice (1 - 4)3
Booting the Surgeon environment...
Writing CBF file.
CBF File Summary:
kernel_crc:	0xfaeceea1
cbf_magic:	0x9abcdef0
cbf_version:	0x00000001
kernel_load:	0x00208000
kernel_jump:	0x00208000
sum_crc:	0x9addb424
size:		0x00616ad0
Compressed: 	 True
Booting surgeon.
Done! Waiting for Surgeon to come up...Done!
root@169.254.8.1's password: 
`

SMOMusic, 2023-02-04:
I am having the exact same issue on Kali. I am not very experienced with Linux so I may be doing something wrong.

andymcca, 2023-02-05:
RSA keys have been deprecated in newer versions of OpenSSH which are supplied with Linux, hence this problem.  Until mac releases a new version of Retroleap which will remove the need for keys, you need to flash using an older version of Linux.

F1atMan, 2023-02-07:
> RSA keys have been deprecated in newer versions of OpenSSH which are supplied with Linux, hence this problem. Until mac releases a new version of Retroleap which will remove the need for keys, you need to flash using an older version of Linux.

Could you give directions with an older version of Linux. Ive been up all night trying to find one compatible.

Chap-With-a-Yellow-Cap, 2023-02-08:
> RSA keys have been deprecated in newer versions of OpenSSH which are supplied with Linux, hence this problem. Until mac releases a new version of Retroleap which will remove the need for keys, you need to flash using an older version of Linux.

Can you specify how old? I tried Bionic Beaver, and had the same problem.

andymcca, 2023-02-10:
That's puzzling - if I get time this weekend I'll see if I can reproduce

jipeoples, 2023-02-11:
I'm having this issue as well on both ParrotOS (ver. 5.1) and Ubuntu (ver 18.04).

andymcca, 2023-02-11:
I'm going to try Slax 9.11 (http://ftp.cvut.cz/slax/Slax-old/Slax-9.x/) - the ISO is from 2019.

Chap-With-a-Yellow-Cap, 2023-02-11:
> I'm going to try Slax 9.11 (http://ftp.cvut.cz/slax/Slax-old/Slax-9.x/) - the ISO is from 2019.

I saw that you forked the correct instructions for getting the keys, but the script that people will get when cloning the repo will not write the keys to the configs (or, at least I think that's what it's doing).

F1atMan, 2023-02-12:
> I saw that you forked the correct instructions for getting the keys, but the script that people will get when cloning the repo will not write the keys to the configs (or, at least I think that's [...]

Anyway to fix this from our end. Or should we wait for an update?

Chap-With-a-Yellow-Cap, 2023-02-12:
> Anyway to fix this from our end. Or should we wait for an update?

I would recommend following the guide until after you've cloned the repo, grab the newest sshflash release's source code and import the "remote flash.sh" file into the folder you cloned the repo into.  There's probably an easier way, but I'm not sure how much the other scripts have changed since that release and now.

andymcca, 2023-02-12:
@Chap-With-a-Yellow-Cap I think doing the git checkout after cloning should put the scripts and keys back to that specified commit i.e. before the changes were made?  When I tested it, I checked the remote_flash.sh script after doing the checkout and it had reverted back to using the keys?

Chap-With-a-Yellow-Cap, 2023-02-12:
@andymcca  My bad, mine didn't, but that might have been because I had a remote_flash already in there.  I'll have to test it again.

andymcca, 2023-02-12:
@Chap-With-a-Yellow-Cap for what it's worth - I just built a new system with Linux Mint 18 Xfce, followed the new instructions on the Retroleap Readme and flashed my Leapster GS without a hitch.  So I think it's all good.

@BullieCaSa can you close this issue now as resolved?

BullieCaSa, 2023-02-14:
> @BullieCaSa can you close this issue now as resolved?

Thanks for the help in getting this resolved!

closed 2023-02-14 by BullieCaSa