mac2612/retroleap #52: LeapsterGS: Power button quick press does not perform graceful shutdown

bmarr, 2023-02-12:
For the LeapsterGS, the  Power button long press works but quick press does not perform graceful shutdown.

However, when connected via ssh, using the poweroff shell command does power off the device.

Somehow the power button doesn't act as a keypress, yet it does work to force a power off when held down long enough.

bmarr, 2023-02-12:
The power button is indeed being registered as a keypress, but the graceful shutdown does not do the expected shutdown.
```
# tail -f /var/log/messages
Mar 13 02:31:49 leapster user.info kernel: lfp100_get_power_button.822 power button status 1
Mar 13 02:31:49 leapster user.info kernel: lf2000-power lf2000-power: Power button pressed
Mar 13 02:31:49 leapster user.info kernel: lf2000-power lf2000-power: shutdown requested

```

andymcca, 2023-02-12:
Can verify this is the case - even though quick press is supposed to shutdown gracefully as you say.  On Retroleap for LF1000 (Gmenunx), a quick press does work as expected i.e. brings up a menu to Power Off, Reboot or Cancel.  But not so in RetroArch unfortunately.

DCFUKSURMOM, 2023-02-14:
might just be able to map the button to run the "poweroff" command
basically the "stfu" of linux commands

DCFUKSURMOM, 2023-02-14:
not as graceful as a fancy menu, but still fine

DCFUKSURMOM, 2023-02-14:
maybe get fancy with it lol, single tap for sleep (if we get suspend working), double for reboot, tripple for shutdown. I wouldnt even know where to begin there

claud9999, 2023-02-15:
> maybe get fancy with it lol, single tap for sleep (if we get suspend working), double for reboot, tripple for shutdown. I wouldnt even know where to begin there

I am having hte same issue. Personally I'd like:
short press => sleep
long press => shutdown (I am guessing this has to be)
home + power => gracious shutdown