mac2612/retroleap #53: Leapster Explorer - Error installing Surgeon

Ma7ya7, 2023-02-15:
Hey, First thank you for the great work you are doing, keep it up.

Trying to install retroleap on my **Leapster Explorer** and I keep getting the same error , I tried  different USB cables, tried on Windows and now trying on Ubuntu. Am I having two bad cables , or the Explorer is bad or not compatible : 

1. LF1000-Didj (Didj with EmeraldBoot)
2. LF1000 (Leapster Explorer)
3. LF2000 (Leapster GS, LeapPad 2, LeapPad Ultra XDI)
4. LF3000 (LeapPad 3, LeapPad Platinum)
Enter choice (1 - 4)2
Writing CBF file.
CBF File Summary:
kernel_crc:	0xd0924a78
cbf_magic:	0x9abcdef0
cbf_version:	0x00000001
kernel_load:	0x00100000
kernel_jump:	0x00100000
sum_crc:	0x9a9e4da8
size:		0x00229354
Compressed: 	 True
Booting the Surgeon environment...
Writing CBF file.
CBF File Summary:
kernel_crc:	0x9b9857a1
cbf_magic:	0x9abcdef0
cbf_version:	0x00000001
kernel_load:	0x00100000
kernel_jump:	0x00100000
sum_crc:	0x9aea2a14
size:		0x0056f4e0
Compressed: 	 True
Traceback (most recent call last):
  File "boot_surgeon.py", line 25, in <module>
    do_surgeon_boot(sys.argv[1])
  File "boot_surgeon.py", line 16, in do_surgeon_boot
    pager_client.upload(path)
  File "/home/mahmoud/sshflash/pager.py", line 114, in upload
    self.rerror(e)
  File "/home/mahmoud/sshflash/pager.py", line 64, in rerror
    assert False, 'Pager Error: %s' % e
**AssertionError: Pager Error: [Errno 2] No such file or directory**
Done! Waiting for Surgeon to come up...Done!
ssh: connect to host 169.254.8.1 port 22: No route to host

Thank you in advance

HighOnVoltage, 2023-02-16:
Got the same issue, pager.py belongs to the [OpenFLConnect](https://github.com/jrspruitt/OpenLFConnect) project, 
I'm running Python 2.7.18, that solved my first error that I found. We could probably add a note to check that you've got python2 installed. 

So I wonder what's changed from the [original requirements](https://github.com/jrspruitt/OpenLFConnect#requirements)? Apprently it was working at some stage or is it an ssh flasher error? 

Traceback (most recent call last):
  File "boot_surgeon.py", line 25, in <module>
    do_surgeon_boot(sys.argv[1])
  File "boot_surgeon.py", line 16, in do_surgeon_boot
    pager_client.upload(path)
  File "/home/tradietrev/Downloads/sshflash/pager.py", line 114, in upload
    self.rerror(e)
  File "/home/tradietrev/Downloads/sshflash/pager.py", line 64, in rerror
    assert False, 'Pager Error: %s' % e
AssertionError: Pager Error: [Errno 2] No such file or directory
Done! Waiting for Surgeon to come up...Done!
ssh: connect to host 169.254.8.1 port 22: No route to host

Ma7ya7, 2023-02-19:
> Got the same issue, pager.py belongs to the [OpenFLConnect](https://github.com/jrspruitt/OpenLFConnect) project, I'm running Python 2.7.18, that solved my first error that I found. We could probably add a note to check that you've got python2 installed.
> 
> So I wonder what's changed from the [original requirements](https://github.com/jrspruitt/OpenLFConnect#requirements)? Apprently it was working at some stage or is it an ssh flasher error?
> 
> Traceback (most recent call last): File "boot_surgeon.py", line 25, in do_surgeon_boot(sys.argv[1]) File "boot_surgeon.py", line 16, in do_surgeon_boot pager_client.upload(path) File "/home/tradietrev/Downloads/sshflash/pager.py", line 114, in upload self.rerror(e) File "/home/tradietrev/Downloads/sshflash/pager.py", line 64, in rerror assert False, 'Pager Error: %s' % e AssertionError: Pager Error: [Errno 2] No such file or directory Done! Waiting for Surgeon to come up...Done! ssh: connect to host 169.254.8.1 port 22: No route to host

I followed all the reuirements , I have python 2.7 installed , sometimes I got different error : 
Pager Error: Mount Error: Device not found.

Aliliche, 2023-02-20:
I got the same issue, the device is not mounted but recognized but the kernel. 
https://askubuntu.com/questions/95606/how-can-i-mount-my-leapfrog-tag-usb-device

verbes4, 2023-06-28:
same issue here, using python 2.7.18

niutech, 2024-01-25:
Are you using sshflash-2.0-alpha4 and retroleap_v2.0.0-alpha10? I've successfully flashed Retroleap on Leapster Explorer with those versions.

Aliliche, 2024-01-25:
@niutech  what is your computer config ?

niutech, 2024-01-25:
Ubuntu 18.04 (bionic)

Jace-198, 2024-06-05:
ATTENTION 
THE SOLUTION TO THE FAILED NETWORK CONNECTION IS TO SWITCH FROM PYTHON 3 TO PYTHON 2