mac2612/retroleap #68: Is it possible to revert to default os?

cornflakes6374, 2023-05-03:
i wonder if its possible to revert to default leapfrog os after installing retroleap? (if you have a backup)

cornflakes6374, 2023-05-03:
btw device is leapster explorer, leapfrog connect app for this device including GS has reached end of life.

andymcca, 2023-05-04:
It's something that's been discussed before - in theory sshflash can do it if you have an appropriate filesystem image of the original OS in each instance,  It's outside the scope of the Retroleap project as it stands but given all the sources etc are available it just needs someone willing, able and with time on their hands(!) to put it together.  Do you agree @mac2612 ?

cornflakes6374, 2023-05-10:
> It's something that's been discussed before - in theory sshflash can do it if you have an appropriate filesystem image of the original OS in each instance, It's outside the scope of the Retroleap [...]

i think i found a url for it:
http://lfccontent.leapfrog.com/lexplorer/downloads/packages/
there is no file listing so i dont know the filenames inside this url

cornflakes6374, 2023-05-31:
someone on the openlf discord server dumped the whole os, so it might be possible to flash it