mac2612/retroleap #71: OhBoy isn't displaying full screen on leapster explorer

comfortabletrain, 2023-05-05:
Hello

I'm currently using v2.0.0 alpha 6 on a leapster explorer and when I launch a Game Boy Color game or a Game Boy game it is only using the center part of the screen, it's roughly the same size as the initial retroleap logo, (although the aspect ratio is correct). When I was using alpha 5 the aspect ratio wasn't right but it was full screen, and much easier to see :).
 
Does anyone have any suggestions to make this full screen?

cornflakes6374, 2023-05-05:
maybe editing ohboy configs with ssh or scp?

comfortabletrain, 2023-05-05:
Sorry, I'm kinda a noob. where are the configs located and how would I would I change them?
Thanks!

cornflakes6374, 2023-05-06:
too lazy to tell but maybe in the ssh shell type
`find / -name ohboy`

cornflakes6374, 2023-05-06:
got it, its at /configs/ohboy

comfortabletrain, 2023-05-09:
Sorry for not responding sooner.
I don't know how to access the ssh shell for the leapster whenever I try to ssh into it, it doesn't ask for a password, it just brings up a line that starts with an arrow. I extracted the configs from the release file that was flashed over, so would it work if I found a working config (e.g. a previous release) and copied the config over, and if that would work then when copying the new file would it replace the old or rename the new?
Thanks for your help.

cornflakes6374, 2023-05-09:
> Sorry for not responding sooner. I don't know how to access the ssh shell for the leapster whenever I try to ssh into it, it doesn't ask for a password, it just brings up a line that starts with an [...]

you need openssh 8.9 as 9.0 is not supported (unless youre on alpha 5 or higher)

comfortabletrain, 2023-05-10:
I don't know what ssh version I have but I am using alpha 6

cornflakes6374, 2023-05-10:
> I don't know what ssh version I have but I am using alpha 6

`ssh -V`

comfortabletrain, 2023-05-10:
Ok, I was able to connect via ssh (I had version 8.4) and I got to the OhBoy configs using vi, however, I may have fried my Leapster somehow so assuming I can revive it, would do I need to change in the config?

- source "Select Palette"
- set palette "Select Palette"
- set colorfilter 0
- set filterdmg 0
- set upscaler 4
- set bmpenabled 0
- set border "Select DMG Border"
- set gbcborder "Select GBC Border
- set red 0x00005f 0x000019 0x000000 0x000023
- set green 0x000019 0x0000aa 0x000019 0x000023
- set blue 0x000019 0x00003c 0x00007d 0x000028
- set blendframes 0

cornflakes6374, 2023-05-10:
hmm i dont know i couldnt find any alpha6 uses the latest ver of ohboy which removes that so you may need to download an old version of ohboy compile it then transfer it to your device with scp

comfortabletrain, 2023-05-12:
Ok, I'll look into that, Thanks for your help!

cornflakes6374, 2023-05-13:
> Ok, I'll look into that, Thanks for your help!

No problem. :)