mac2612/retroleap #76: Is there any other console supports retroleap other than LeapsterGS and Leappad2?

Crazysheepexe, 2023-05-29:
I have an old Leapster2, found in a pawn shop, fixed it and it is working. I bought it because for this, but I see that it may not support retroleap, and I want (in anytime) to have it happen, even if I have to get involved (idk if I can) to make it happen.
Can you tell me when it may support Leapster2 or the others, or at least just how did you make it supported for leapsterGS and Leappad2?
Thank you.

heyitzrare, 2023-05-29:
**Absolutely never happening.** The supported devices all run Linux, and can be modified, while the Leapster2 is a Flash device - supporting the Leapster2 would mean completely rewriting literally everything for a single device. To my knowledge, homebrew *does* exist for the Leapster2, but RetroLeap will *never* support the Leapster2.

Crazysheepexe, 2023-05-31:
There's no such thing as "never happening". Like what I've said, I will do anything for it, even if I have to break the leapster2's system piece by piece to make it run linux, same goes to the program or the system I will put into. And also I've asked, what can I do or what thing should I start with to make things go to work so it makes Leapster2 support linux.
And that's all.
Thank you for the answer.

heyitzrare, 2023-05-31:
In theory, you may be right, but "in theory" and "in practice" are two completely separate things. If you manage to get RetroLeap onto a Leapster2, and make it even slightly playable, I and many others will literally hail you as the god of L2 hacking, mark my words

Crazysheepexe, 2023-06-12:
Wish me luck then 😉

heyitzrare, 2023-08-21:
So, any luck?