mac2612/retroleap #77: add leaptv support

cornflakes6374, 2023-06-01:
the leaptv has 16gb storage, 1ghz quad core cpu, and 1gb ram, which is very superior hardware. it is a hackable system (someone ran doom on it) and it would be cool to add leaptv support but sadly there is no leapfrog connect app for leaptv so this might require soldering or modifying apps to run emulators or retroarch. you can find some pages in spiffyhacks on wayback machine for leaptv hacking tutorials. so i wonder is it possible to add support for leaptv? (or maybe not?)

trilo2600, 2023-06-01:
If it runs Linux I’m sure it can be hacked!Sent from my iPhone

cornflakes6374, 2023-06-02:
yep, it does run linux on the internal os where there is a secret serial port on there which the linux console can be accessed.

heyitzrare, 2023-08-21:
Entirely possible, although it'd be a pretty difficult task. If I find a LeapTV at a thrift store I'll 100% join in testing lol

cornflakes6374, 2023-08-26:
I dont have a leaptv yet and im probably not a good programmer so i hope mac does it