mac2612/retroleap #84: SD Booting?

mistermuki, 2023-09-27:
Hello!

I had a private project I was working on that was based on the original LeapFrog source code and I was able to get a reliable SD booting using a custom cartridge I developed. It required a bootloader patch (a version of it is documented on the e-linux forums I believe).

I was wondering what the ethos of this project was in terms of running Retroleap off an microSD card and leaving the core OS untouched? Seems like it would be better in general especially with system restore being spotty recently.

heyitzrare, 2023-09-27:
If this bootloader patch is completely undetectable by the main OS, then what you've just done is create dual-boot for the Explorers, and we might as well call you a god. Although I would highly recommend not using LF source code if you don't want ninja frogs busting down your door lol

trilo2600, 2023-09-27:
Oh that’s a really cool idea!How large of an sd card can you boot from that custom cart and is the sd card removable?Sent from my iPhone

mistermuki, 2023-09-27:
I think I used a 16 GB card the last time I did this. It's been over a year since the last time I touched it but it worked pretty reliably. I can explore bringing that functionality over to here.

Here it is running DOOM off of the SD card. You can see the custom cartridge at the top of the frame.
[image]

mistermuki, 2023-09-27:
It should, they use the same cartridge. I can see about bringing the same SD functionality over to retroleap and testing on GS can be a part of that.

bladehorton, 2024-04-21:
I'm assuming this bypasses the main OS and takes over as the primary boot device in the bootloader. In that case, does the main OS still boot without the cartridge inserted, or maybe a button combination at boot to switch to the main OS? How would the bootloader react to a standard game cartridge, not your SD card?

mistermuki, 2024-05-20:
This repo seems to be very slow moving so I'm going to be doing a lot more of my SD card based work over at the repo over [here](https://github.com/leapbunny). A couple things are already over there (like custom cartridges).

mistermuki, 2024-06-03:
> I'm assuming this bypasses the main OS and takes over as the primary boot device in the bootloader. In that case, does the main OS still boot without the cartridge inserted, or maybe a button [...]

@bladehorton It still boots the default OS if it doesn't find an SD card. It should play games normally (but I don't have any to test) because they won't have a valid kernel which it checks for.