mac2612/retroleap #89: root@169.254.8.1's password:  flashing

foley-310, 2024-01-10:
when trying to flash retroleap oc to my leapstergs from ubuntu i get a prompt which requires a password to progress any further and i dont have a password. 

[image]
i am running ubuntu 22.04.03 lts from a raspberry pi 4b

niutech, 2024-01-25:
Are you using sshflash-2.0-alpha4? It's the last version to use SSH keys to log in to Leapster.

foley-310, 2024-02-12:
> Are you using sshflash-2.0-alpha4? It's the last version to use SSH keys to log in to Leapster.

i just copy and pasted the command for ssh from this github into my terminal and am using that

wibwib, 2024-04-06:
I am also getting this issue and am using sshflash-2.0-alpha4.

BigBankBailey, 2024-11-07:
Same two years of trying on Ubuntu pop os and windows still asks for public key password bs