mac2612/retroleap #97: root@169.254.8.1's password

wibwib, 2024-04-06:
So essentially when trying to install the overclocked version of RetroLeap on my leapster GS (using ubuntu,) I get this message saying i need  to enter the password. Ive seen this issue many times in this issue section, but none of them seem to help. I am using directly the version of SSHFLASH that was provided by mac2612. Is there anything i can do? I am very new to linux and havent used this essentially ever.

brodieman23, 2024-10-17:
Yeah I have the same issue. I tried copying over the key from the backup but nope. Some additional documentation would be great. All I find is a video of Bringus trying, seeing the error and then installing with the windows version (which I cannot use)

KevDoy, 2025-02-16:
Did you indicate the ssh key?