mac2612/retroleap #98: Cannot remove 'surgeon_tmp.cbf'

Deadman422, 2024-04-21:
Hi,

I'm using Linux Mint 21.2 and a Leapster Explorer. I've downloaded Python 2 and Pager, and the console itself recognizes that it's plugged in.  My system gives the same output weather the Leapster is plugged in or not. I am using retroleap v1.2.1 and sshflash 2.0. What am I doing wrong? 

Here's what it comes up with:

1. LF1000-Didj (Didj with EmeraldBoot)
2. LF1000 (Leapster Explorer)
3. LF2000 (Leapster GS, LeapPad 2, LeapPad Ultra XDI)
4. LF3000 (LeapPad 3, LeapPad Platinum)
Enter choice (1 - 4)2
Traceback (most recent call last):
  File "make_cbf.py", line 10, in <module>
    cbf.create(mem=sys.argv[1], opath=sys.argv[3], ipath=sys.argv[2])
  File "/home/user/sshflash/cbf.py", line 142, in create
    error(e)
  File "/home/user/sshflash/cbf.py", line 60, in error
    assert False, e
AssertionError: Path does not exist.
Booting the Surgeon environment...
Traceback (most recent call last):
  File "make_cbf.py", line 10, in <module>
    cbf.create(mem=sys.argv[1], opath=sys.argv[3], ipath=sys.argv[2])
  File "/home/user/sshflash/cbf.py", line 142, in create
    error(e)
  File "/home/user/sshflash/cbf.py", line 60, in error
    assert False, e
AssertionError: Path does not exist.
[sudo] password for user:            
Traceback (most recent call last):
  File "boot_surgeon.py", line 25, in <module>
    do_surgeon_boot(sys.argv[1])
  File "boot_surgeon.py", line 16, in do_surgeon_boot
    pager_client.upload(path)
  File "/home/user/sshflash/pager.py", line 114, in upload
    self.rerror(e)
  File "/home/user/sshflash/pager.py", line 64, in rerror
    assert False, 'Pager Error: %s' % e
AssertionError: Pager Error: Surgeon not found.
Done! Waiting for Surgeon to come up...rm: cannot remove 'surgeon_tmp.cbf': No such file or directory
Done!
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Detected Kernel partition= RFS Partition=
Flashing the kernel...ssh: connect to host 169.254.8.1 port 22: No route to host
cat: zImage_tmp.cbf: No such file or directory
ssh: connect to host 169.254.8.1 port 22: No route to host
Done flashing the kernel!
Flashing the root filesystem...ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Writing rootfs image...
cat: lf1000_rootfs.tar.gz: No such file or directory
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Done flashing the root filesystem!
Do you want to format the roms partition? (You should do this on the first flash of retroleap) (y/n)y
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
ssh: connect to host 169.254.8.1 port 22: No route to host
Done! Rebooting the host.
ssh: connect to host 169.254.8.1 port 22: No route to host

Nothing appears on the screen of the console while this happens. 

here is the dmesg: https://pastebin.com/v5kTqJVZ

eddiefresh, 2024-06-07:
I’ve gotten this message only when trying to installed the overclocked version for the Leapster GS, as long as you follow the instructions correctly it should work, I got it working using Ubuntu 18.4
[image]

eddiefresh, 2024-06-07:
[image]

Espiano973, 2024-08-01:
I'm getting this to but on the leappad 3 I'm using pop os with is Ubuntu based on oracle vm in another issue it said you need to download it with sudo pip install pager cbf but it doesn't work also I'm on windows 11

Espiano973, 2024-08-24:
I got it to work using vm ware

blackletum, 2024-09-06:
@Deadman422 I am getting the exact same errors as you on two different machines.

Did you ever get this working?

I am attempting to get this working on a Leapster GS with no success

blackletum, 2024-09-06:
> I got it to work using vm ware

what OS were you using in vmware?

Espiano973, 2024-09-09:
Pop os

Espiano973, 2024-09-09:
You can download here https://pop.system76.com/

Alfa-PT, 2025-04-05:
Hi guys!. . . Any luck on trying to install the OC version? Having issues here, already tried Ubuntu, Mint, Windows and such won't work. same issue, connection timed out and port 22 error . any luck ? ( Trying now Vmware )